RaspberryPiFoundation/python-build-hat #96: buildHat unusable

radu022003, 2021-11-09:
Hello,
after installing the board and the python library i can't create successfully any object (Motor or anything else).
when trying to create a motor, with the motor connected, but no 8V available i get the following error:
```
Traceback (most recent call last):
  File "/home/pi/python_script/lego2.py", line 4, in <module>
    mot=Motor('A')
  File "/home/pi/.local/lib/python3.9/site-packages/buildhat/motors.py", line 27, in __init__
    raise DeviceInvalid('There is not a motor connected to port %s (Found %s)' % (port, self.name))
buildhat.exc.DeviceInvalid: There is not a motor connected to port A (Found No device)
```
I can observe that the red led on the board is turning off for short time and than back on.

I have also connected to the board using minicom and i can see the followings:

`BHBL> help
BuildHAT bootloader version 1631704133 2021-09-15T12:08:53+01:00
Commands available:
  help, ?                  : display this text
  version                  : display version string
  load <length> <checksum> : initiate upload
                             followed by <STX><<length> data bytes><ETX>
  signature <length>       : upload signature
                             followed by <STX><<length> data bytes><ETX>
  clear                    : clear any uploaded image
  verify                   : verify upload
  reboot                   : reboot if upload successfully verified
BHBL>
`
According to documentation, there should be a lot more commands available, which leads me to the ideea that the firmware update is not succeeding.

Can you please give a hint how to make this work?
Thank you and best regards

chrisruk, 2021-11-09:
Hi, I'm just wondering how you're powering the Pi, are you using a Raspberry Pi power supply connected to the usb-c port if you're not using the barrel jack connector on the buildhat itself?

radu022003, 2021-11-09:
maybe i have should also say that i am using it with a raspberry pi 3B+, but for my issues i guess there is no difference.
I have used both methods, supplying only the raspberry pi, over the  microUSB cable and another try by powering both (pi and buildHat) over the jack connector.
i have debugged a bit the communication during the firmware update of the buildHat:

```
firmware loaded
threads started
rebooted after firmware loaded
data from board:

data from board:  Image received

data from board:  Checksum OK

data from board:  BHBL>

data from board:  BHBL> signature 64

data from board:

data from board:  Signature received

data from board:  BHBL>

data from board:  BHBL> reboot

data from board:  Verifying image...

data from board:  Verifying image length=49416 21 21 00 20

data from board:  SHA256:

data from board:   5B 73 34 12 81 94 CA 57 65 BC FA 54 50 BF 76 40 2F 41 01 44 6D 5A 96 AA 6C 6E 10 EA C4 FA 2D 68

data from board:  Public key:

data from board:   40 58 70 58 43 39 16 4F 9D EA DF 8B B7 16 73 5C 53 1E 3A CC 0C E3 9E 4E 29 12 C0 69 07 68 DF 5F D1 B4 C5 0A D3 97 3D D1 B2 75 29 4E 58 82 6D 8A E1 99 C0 A0 7F FE 07 9D 6C 6E 1D 7D BE A9 3A F9

data from board:  Image verifed OK

data from board:  Rebooting...

data from board:

data from board:  Type help <RETURN> for help

data from board:  Initialising ports

data from board:  Resetting drivers: 0123

data from board:  Reading driver dumps: before port_initdriver(): 0123

data from board:  Reading driver dumps: after port_initdriver(): 0123

data from board:  Done initialising ports

data from board:
```
I can see that the firmware and signatures are uploaded correctly and even after the reboot, the ports are correct initialized, but when i am connecting with minicom to buildHat, i see the bootloader again.

radu022003, 2021-11-10:
I have also tested the board on a raspberry pi 4, there everything works as expected, no errors.
What could be wrong with raspberry pi 3, i have a fresh OS installed

topshed, 2021-11-10:
Are you using the same SD card on the Pi 4 as the 3B+?

radu022003, 2021-11-10:
> Are you using the same SD card on the Pi 4 as the 3B+?

No i am not using the same SD card for Pi 4 or 3B+.
But doing more tests with the 3B+ i could observe that by activating the 1-wire interface, with raspi-config, leads to this behaviour. As soon as the 1-wire interface is deactivated, everything is back to normal.
I want into this problem just because i want to use a DS18B20 together with the buildHat, therefore I activated it from the very beginning.
It seems that SPI or I2C is not disturbing the buildHat at all.

I will try the same on RPi 4 too and get back.

topshed, 2021-11-10:
GPIO pin 4 (the default for 1-wire)  is used by the Build HAT. You can select a different pin for 1-wire by adding a dtoverlay to /boot/config.txt. For example, to use pin 23 instead:

`dtoverlay=w1-gpio,gpiopin=23`

If that works, please let us know.

radu022003, 2021-11-10:
In between i have only tested on RPi4 (without the proposed change) and the problem is the same.
I will try it out and let you know.

radu022003, 2021-11-10:
> GPIO pin 4 (the default for 1-wire) is used by the Build HAT. You can select a different pin for 1-wire by adding a dtoverlay to /boot/config.txt. For example, to use pin 23 instead:
> 
> `dtoverlay=w1-gpio,gpiopin=23`
> 
> If that works, please let us know.

i confirm that it works on RPi 3B+

closed 2021-11-10 by topshed