RaspberryPiFoundation/python-build-hat #102: Please provide a way to specify a different serial device to use

Parlane, 2021-12-06:
The following line limits the serial device to talk to; to serial 0, but on my ubuntu rpi device serial0 is taken by the system and so the buildhat comes up as serial1, to use this library I have to manually edit the source of serinterface.py on my system to use serial1. Thank you.

https://github.com/RaspberryPiFoundation/python-build-hat/blob/d2d6ff3f43de8a92ef8c900d1b278f1298a29c0a/buildhat/serinterface.py#L60

chrisruk, 2021-12-13:
Hi, I've just added support for using an alternative serial device to main of this repo, I will get it added to the next pypi release.  You can test it now like so though:

```
pip3 uninstall buildhat
pip3 install git+https://github.com/RaspberryPiFoundation/python-build-hat.git
```

Then in your script:

```
from buildhat import Hat, Motor
hat = Hat(device="/dev/serial1") # must be at start
motor = Motor('A')
motor.run_for_degrees(180)
```

closed 2021-12-17 by chrisruk

droneconnect, 2021-12-20:
does this mean we can use 2 now hats stacked and drive up to 8 motors using 2 serial ports?

chrisruk, 2021-12-21:
@droneconnect Sorry unfortunately not.  It just means you can choose which serial port to use rather than it always being /dev/serial0

caco3, 2023-11-15:
> Hi, I've just added support for using an alternative serial device to main of this repo, I will get it added to the next pypi release. You can test it now like so though:

Thanks!
I suggest to add this to the documentation as well. I only found it by searching for this issue.

aallan, 2023-11-15:
> I suggest to add this to the documentation as well.

We take PRs. https://github.com/raspberrypi/documentation/tree/develop/documentation/asciidoc/accessories/build-hat