RaspberryPiFoundation/python-build-hat #105: Support for LEGO 88005 Powered Up Light

Niwa1980, 2021-12-11:
Please add support for LEGO 88005 Powered Up Light.

https://www.lego.com/en-us/product/light-88005

chrisruk, 2021-12-13:
Hi, I've just added support for the light, I will get it added to the next pypi release.  You can test it now like so though:

```
pip3 uninstall buildhat
pip3 install git+https://github.com/RaspberryPiFoundation/python-build-hat.git
```

Then in your script:

```
import time
from buildhat import Light
light = Light("D")
light.on()
time.sleep(1)
light.off()
```

chrisruk, 2021-12-21:
Hi,

Just wondering if you're still having problems with this?

I've just been testing turning on/off the light on a clean boot
and also after unplugging / plugging in the light after.

Thanks

Niwa1980, 2021-12-23:
Hi,

sorry for the delay. I checked the latest version with no success. This code:

```
from buildhat import Hat
hat = Hat()
devices = hat.get()
print(devices)
```

produces the following output: 

```
{'A': {'typeid': -1, 'connected': False, 'name': 'Disconnected'}, 'B': {'typeid': -1, 'connected': False, 'name': 'Disconnected'}, 'C': {'typeid': -1, 'connected': False, 'name': 'Disconnected'}, 'D': {'typeid': -1, 'connected': False, 'name': 'Disconnected'}}
```

It is irrelevant at which port the leds are connected. There are not recognized.

I will check now if the leds are working properly.

Cheers

Niwa1980, 2021-12-23:
I ordered a replacement. After delivery I will test again.

mutesplash, 2022-01-14:
Light support doesn't work.  This seems to be a firmware problem.  I plugged it into the 51515 hub and was able to set the light levels from 0-10, so the light is not broken.

mutesplash, 2022-01-15:
The **old** Firmware 1630165527 2021-08-28T16:45:27+01:00 will power on the lights, but not list them in the device list.

joerk, 2022-01-16:
Upgrade to 0.5.7 release but the Light 88005 on port D is not detected. 

`pi@pi4:~/projects/lego_buildhat $ /bin/python /home/pi/projects/lego_buildhat/hat.py
{'A': {'typeid': 75, 'connected': True, 'name': 'Motor'}, 'B': {'typeid': 63, 'connected': True, 'name': 'ForceSensor'}, 'C': {'typeid': 62, 'connected': True, 'name': 'DistanceSensor'}, 'D': {'typeid': -1, 'connected': False, 'name': 'Disconnected'}}`

Any hints?

chrisruk, 2022-01-17:
Could everyone tell me what PSU you're using please with the buildhat? I had issues with the lights when using USB-C PSU, but with my own barrel jack PSU, they do work for me.

Niwa1980, 2022-01-17:
I am using the official PSU: https://www.raspberrypi.com/products/build-hat-power-supply/.

chrisruk, 2022-01-17:
Thanks, do you mind telling me where you bought the light from? (Just wondering if there maybe more than one version)

Niwa1980, 2022-01-17:
From the official Lego online shop.

Niwa1980, 2022-01-17:
Lego support tells me, that the product will be available at the end of january. I ordered a replacement in december 2021 for the supposedly defect component.

mutesplash, 2022-01-17:
Official 8V 6A RPi PSU

mutesplash, 2022-01-17:
Also bought the light from lego.com

chrisruk, 2022-01-17:
Thanks all, just ordered some from lego.com to have a look at

joerk, 2022-01-17:
> Thanks, do you mind telling me where you bought the light from? (Just wondering if there is maybe more than one version) I will see if I could possibly order one from the seller someone here used, to do some more debugging.

From the local lego store - cologne; germany. See attached the bag ... 
[image]
[image]
[image]

joerk, 2022-01-17:
> Official 8V 6A RPi PSU

> I am using the official PSU: https://www.raspberrypi.com/products/build-hat-power-supply/.

Same here.

Niwa1980, 2022-02-08:
Any news on this?

mutesplash, 2022-02-14:
`P0: connected to passive ID 8`

Shows up and works with the Light class!

Can you set the brightness levels like you can on the Spike hub?

chrisruk, 2022-02-15:
@mutesplash sure, just added a brightness function:

```
import time
from buildhat import Light

l = Light('A')
bright = 0
step = 1
while True:
    l.brightness(bright)
    bright += step
    if bright == 100 or bright == 0:
        step *= -1
    time.sleep(0.01)
```

It uses 0 - 100 for consistency with the brightness on the distance sensor.

Niwa1980, 2022-02-15:
I will try to test it tomorrow.

Thanks in advance.

Niwa1980, 2022-02-16:
It's working. Great work and thanks again. 👍

closed 2022-02-16 by Niwa1980

joerk, 2022-02-16:
> @Niwa1980 @joerk Does this work for you? Great success over here!

Yes! Take me some time to setup the local build env. Now I'm prepared for deep dive ;-) Regards Joerk