RaspberryPiFoundation/python-build-hat #106: Error message when running serial command

Narshil94, 2021-12-24:
I followed the getting started document and ran the recommended python script. The first time i got a response along the lines of no motor connected on port A. Now it runs the script with no output at all.

I tried connecting to the board directly through serial0.

The list command returns Error.

The version command responds with BuildHAT bootloader version 1631704133 2021-09-15T12:08:53+01:00

Is there somewhere where I can find a more recent bootloader version?

chrisruk, 2022-01-04:
Can you show me which python script you're running, along with the behaviour you're seeing with the script

Thanks

chrisruk, 2022-01-26:
Closing for now, please re-open if you're still facing an issue

closed 2022-01-26 by chrisruk

zer044, 2022-07-23:
Just in-case its useful to anybody else. The issue was creating the image on a RPI3B+ then running the image on a RPI3 A+. This does not work and for some reason remains broken when you move the sd card back to a RPI3B+. 

Running on a RPI3B+ without moving it to another RPI works fine.