RaspberryPiFoundation/python-build-hat #107: Get voltage info

iwanimsand, 2021-12-29:
The command `vin` exists in the serial protocol [Raspberry Pi Build HAT Serial Protocol](https://datasheets.raspberrypi.com/build-hat/build-hat-serial-protocol.pdf) and prints the voltage present on the input power jack.

When I work with a battery, I want to constantly check the voltage and thus protect against a deep discharge.

I haven't found a method in the official python library to get this information. Would it be possible to implement such a method in the python library?

chrisruk, 2022-01-04:
Yup, that sounds a good plan, will have a look at your pull request soon, thanks!

chrisruk, 2022-01-10:
Thanks for your pull request, I've just merged it

closed 2022-01-10 by chrisruk