RaspberryPiFoundation/python-build-hat #118: PassiveMotor: speed problems

porchy13, 2022-01-22:
Hi,

Do you plan to handle the [Wedo 2.0 medium motor](https://www.bricklink.com/v2/catalog/catalogitem.page?S=45303-1&name=WeDo%202.0%20Medium%20Motor&category=%5BEducational%20&%20Dacta%5D%5BWeDo%5D#T=S&O={%22iconly%22:0}) ?

I have defined it as an `PassiveMotor`, and it runs. The only problem I have is the speed. But I also experience these speed problem with a train motor : from a speed of 50 (absolute), it seems that it reachs its maximum speed. I don't know if it is a limitation of the motors or something related to the firmware.

Regards,

Porchy

chrisruk, 2022-01-25:
Hi,

Yep, I need to add the ID for that motor to the list of passive motors.

Passive motors don't have an encoder to measure speed/position like the other motors, so we can't regulate 
the speed in the same way.

Hope that answers your questions

Thanks

porchy13, 2022-01-26:
Thanks for your answer. here is a reformulation about the speed:

The passive motor rotate according to the power (or percent of powerage) that have been set. Using the [PoweredUp Hub 4](https://www.lego.com/product/hub-88009), setting a power of 30 make the medium motor rotate for about 3.5 rotations. Set to 100 (full power), the medium motor rotate for about  12 rotations (tested visualy, not using scientific tools).

I have excluded the Wedo 2.0 Hub from my tests, as it works on low voltage (3V) and, according to the [WeDo 2.0 Communication Software Developer Kit](https://education.lego.com/en-us/product-resources/wedo-2/downloads/developer-kits), an offset acting on the power is configured for the medium motor.

While using the Build Hat, setting a power of 30 or 100 has the same effect, the medium motor rotate at full speed (about 16 rotations, the batteries of my Hub where not fully charged). I also tried with a train motor. With the same power value, it rotates way faster when attached to the Build Hat, than on the PoweredUp Hub. It reaches the maximum speed very quickly.

Trought the PoweredUp Hub or the Wedo 2.0 Hub (only for the medium motor), stepping from 10 to 10, the rotation speed is more marked. At 10, the rotation is very slow. I have never succeed to reached such a slow rotation with the Build Hat.

In conclusion, there is something handled by the Hubs about those motors that is not done on the Build Hat. As the Build Hat has been made in collaboration with LEGO Education, it would be great if they could explain how they get those motors working.

chrisruk, 2022-02-17:
Hi,

Sorry for the delay, I've just added the passive ID for this motor to the list.

I see what you mean about some values having the same effect in speed.

I've just been experimenting more with the train motor.  

I've been trying the following, the speed seems to reach 
the maximum speed of the motor around 80 for me (only judging by the changes
in the audible frequency the motor makes).

```
import time
from buildhat import PassiveMotor
p = PassiveMotor('A')
p.plimit(1.0)
p.bias(0.15)

for speed in range(10, 100+1, 10):
    print("speed ", speed)
    p.start(speed)
    time.sleep(2)

p.stop()
```

chrisruk, 2022-03-03:
Closing for now, please re-open if necessary

closed 2022-03-03 by chrisruk