RaspberryPiFoundation/python-build-hat #122: buildhat.exc.DeviceNotFound: No device found

OliverFaust, 2022-02-25:
Hi,

Writing to the device in a hard loop causes a `No device found' after some time. Here is the complete call trace:
> Traceback (most recent call last):
>   File "/home/pi/Python/xBoxController/minimalError.py", line 5, in <module>
>     pair.start(0,0)
>   File "/usr/local/lib/python3.9/dist-packages/buildhat/motors.py", line 390, in start
>     self._rightmotor.start(speedr)
>   File "/usr/local/lib/python3.9/dist-packages/buildhat/motors.py", line 229, in start
>     self._write(cmd)
>   File "/usr/local/lib/python3.9/dist-packages/buildhat/devices.py", line 155, in _write
>     self.isconnected()
>   File "/usr/local/lib/python3.9/dist-packages/buildhat/devices.py", line 91, in isconnected
>     raise DeviceNotFound("No device found")
> buildhat.exc.DeviceNotFound: No device found

The time delay, after which that error occurs, seems to be random. I have observed that error anywhere from 1 min to 5 min. That might indicate a runtime problem with the framework, such as a buffer overrun. 
Here is a minimal example to reproduce the error:
```
import buildhat

pair = buildhat.MotorPair('A', 'B')
while True:
    pair.start(0,0)
```
Debugging revealed that BuildHAT.DEVTIMEOUT followed by BuildHAT.DISCONNECTED occurred in serinterface.py

mutesplash, 2022-02-25:
Firmware on the RP2040 basically can't handle what you're asking it to do.

OliverFaust, 2022-02-25:
Dear mutesplash,

Thank you for your response. 
Are you going to fix this issue?
From my experiments, I know that the error is correlated with the call frequency to your interface, i.e. if a lower call frequency prolongs the time without error. For example: 
```
import buildhat
from time import sleep
pair = buildhat.MotorPair('A', 'B')
while True:
    pair.start(0,0)
    sleep(0.2)
```
However, there are at least three issues with this workaround:
1. The error still occurs - the program just runs for longer. It is not clear how long to sleep. 
2. The program is less reactive - bad for real time
3. Sleeping is the wrong solution. Blocking is the correct solution: the call, in this case `pair.start(0,0)`, should only return when the framework is ready to take the next command (API call).

mutesplash, 2022-02-25:
Unfortunately there is no way to block in the Python library because the firmware does not respond in the affirmative when you issue a command.

mutesplash, 2022-02-26:
The python library in this repo doesn't seem cover the BuildHAT firmware source change that would create an out-of-band feedback loop that you suggest.  It would be nice if it did, though!  I see where you're coming from and would suggest coming up with a more realistic test case to argue for this change. (I am not a committer)

OliverFaust, 2022-02-26:
Dear mutesplash,

Here is a minimal use case for controlling two motors with a game pad:
```
from xbox360controller import Xbox360Controller
import buildhat
controller = Xbox360Controller(0, axis_threshold=0.2)
pair = buildhat.MotorPair('A', 'B')
while True:
    pair.start(controller.axis_r.x*100, controller.axis_r.y*100)
```
The importance of this test case becomes clearer when we start to use parallel processing. Here is an example an example with two communicating sequential processes that form a process network with the same functionality:
```
from pycsp.parallel import *
from xbox360controller import Xbox360Controller
import buildhat

class xboxBuildHat:
    def __init__(self):
        ch = Channel()
        Parallel(
            self.sampleRightStick(-ch),
            self.MotorProcess(+ch)
            )
    
    @process
    def MotorProcess(self, cin):
        pair = buildhat.MotorPair('A', 'B')
        while True:
            speed = cin() 
            pair.start(speed[0]*100,speed[1]*100)
                                  
    @process
    def sampleRightStick(self, cout):
        controller = Xbox360Controller(0, axis_threshold=0.2)
        while True:
            cout([controller.axis_r.x, controller.axis_r.y])
    
if __name__ == "__main__":
    while True:
        xboxBuildHat()
```
By now pycsp is a bit dated and the library code must be patched for python3, but the example shows the principle of building reactive process networks with blocking channels.

mutesplash, 2022-02-26:
Hm, I see. Without a patch to the closed-source firmware, the BuildHAT library would have to guess at a max input rate and throttle all write()s in the BuildHAT class.  But, it seems you haven't been able to determine what that rate is.

tenderandshy, 2022-03-01:
When I tried to download python 3 library for build hat it didn't work
raise Hat not found()
build hat.exc. Hat Not Found
I am trying to use the build hat with Lego Spike Prime and i am not getting anywhere  Could someone help me, please?

mutesplash, 2022-03-01:
You should use the RPi forums for this kind of support, not github issues, @tenderandshy 

https://forums.raspberrypi.com/viewforum.php?f=45

chrisruk, 2022-03-03:
@OliverFaust I think I see what you're trying to achieve with regards to using a gamepad and controlling the motor speed.

Mutesplash mentioned some good points re. throttling the messages.  I'll have a think about this.

OliverFaust, 2022-03-04:
Unfortunately, there is no official documentation available. However, from what I understand, the H-bridge, which determines the motor speed, sits on the BuildHAT PCB. If this is the case, it might be possible to set the motor speed without communicating over the motor data interface.

chrisruk, 2022-03-18:
Closing now, thanks to @mutesplash for the pull request!

closed 2022-03-18 by chrisruk

rjboonstra, 2022-12-24:
I am seeing a similar issue...  I am trying to use ten (10x) GPIO points to display the motion sensor status; I know there are some issues with this code as the motion sensor values are out as 1=close and 10=far away... but my problem is that the program locks up the Raspberry Pi with the command _GPIO.setup(i, GPIO.OUT)_.  

Just a quick general question to confirm whether or not it is possible to use the GPIO library at the same time as the BuildHat library.  Note I was careful to make sure that I did not use the same GPIO pins that the Build Hat uses so there would not be any interference.  Note that I am also using GPIO 24 for fan control, as the Raspberry Pi 4 runs hot.
_#!/usr/bin/python3

# IMPORT LIBRARIES %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
import time
import RPi.GPIO as GPIO
from buildhat import Motor
from buildhat import MotionSensor

# SET UP IO PORTS ON BUILD HAT %%%%%%%%%%%%%%%%%%%%%%%
motor = Motor('C')
sMotion = MotionSensor('D')
print("Build Hat Varialbes set...")

# SET UP ARRAY FOR LEDS %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
led = [4, 10, 9, 8, 11, 7, 5, 12, 6, 13]
print("Array configured...")

# SET UP GPIO %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
GPIO.setmode(GPIO.BCM)

for i in led:
	GPIO.setup(i, GPIO.OUT)

# INTRODUCTION - TEST LEDS %%%%%%%%%%%%%%%%%%%%%%%%%%%
for i in led:
	GPIO.output(i, GPIO.HIGH)
	time.sleep(0.1)
	print(i)
else:
	time.sleep(.3)
for i in reversed(led):
	GPIO.output(i, GPIO.LOW)
	time.sleep(0.1)

while True:
	iDist = sMotion.get_distance()
	print(iDist)
	if iDist == 0:
		iTime = iTime - 1
		iDist = iDistLast
	else:
		iTime = 20
		iDistLast = iDist
	for i in led:
		j = j +1
		if iDist < j:
			GPIO.output(i, GPIO.HIGH)
		else:
			GPIO.output(i, GPIO.LOW)
	iSpeed = (10 - iDist ) * 10
	motor.start(iSpeed)
	time.sleep(0.3)

GPIO.cleanup()_
[LegoBuildHat Schematic.pdf](https://github.com/RaspberryPiFoundation/python-build-hat/files/10298016/LegoBuildHat.Schematic.pdf)
[LegoBuildHat PCB.pdf](https://github.com/RaspberryPiFoundation/python-build-hat/files/10298017/LegoBuildHat.PCB.pdf)