RaspberryPiFoundation/python-build-hat #123: Ports seem to add, plimit commands apply to previously selected ports

nathankellenicki, 2022-03-03:
Hi,

I'm not using the library but the raw serial interface.

I have a train motor (id 2), a light (id 8), and a Technic large motor (id 46) attached to ports 1, 2, and 3 respectively (Nothing on port 0).

When I issue this command:

`port 1; on; plimit 0.5;`

The train motor moves at half speed. I then stop it.

`port 1; on; plimit 0;`

I then set the lights to half brightness:

`port 2; on; plimit 0.5;`

And the lights start at half brightness, but also the train motor starts moving at half speed again.

I turn off the light and both turn off.

`port 2; on; plimit 0;`

Next I start the Technic motor:

`port 3; on; plimit 0.5;`

And all three devices start at half speed/brightness.

Is this a bug? I would have thought the plimit command would apply to only the port I have selected.

Thanks!

aallan, 2022-03-03:
This sounds more like an issue for the forums, https://forums.raspberrypi.com/viewforum.php?f=45, than this repo.

mutesplash, 2022-03-03:
It is a forum issue, but if you read the serial documentation, the plimit command is global.  The Spike PRIME hub does the exact same thing when a motor and the lights are plugged in without a program running.  This seems to be because of how the motor driver chip works and because the lights are passive.  Although I would think since they're paired, you could at least get separation between two specific pairs of ports, but that might be more complicated to expose.

nathankellenicki, 2022-03-03:
I see, thanks for the replies. Actually looking closer at the Python library I notice the use of pwm and set, I guess this is more correct?

```port 0; pwm; set 1;``` (100% speed)

Which I'll have to try instead. Thanks!

nathankellenicki, 2022-03-04:
Closing as solved!

closed 2022-03-04 by nathankellenicki