RaspberryPiFoundation/python-build-hat #135: motor - run until stalled

KusneNIST, 2022-03-25:
Hi, will the package eventually support motors stopping when stalled, similar to the pybricks functionality? Or is this not possible with the HAT? Is there a way to implement this with the current python-build-hat library?
Thank you!

KusneNIST, 2022-03-25:
My scenario is: having the motor move and once the motor reaches an obstruction and stalls, the motor stops.
This reduces the need for combining a motor with a sensor that detects when the motor has stopped.
Since the HAT has only 4 LEGO ports, this reduces the number of LEGO devices needed for motion control.
Alternatively we're looking at combining two LEGO HATS through a stack board, but there are pin address collisions. So we're also looking at ways to combine two HATS with the ability to multiplex between them.
Thanks again!

mutesplash, 2022-03-31:
I tried doing a stupid simple solution for this and got halfway through and realized it wasn't going to work due to weirdness.  I went and looked at pybricks and their code is amazingly complicated (and excellently documented) and has the benefit of working directly on the motor and its signals.  All this is to say I wouldn't get your hopes up, but I'm not done poking at it.

laurensvalk, 2022-07-26:
> Hi, will the package eventually support motors stopping when stalled, similar to the pybricks functionality? Or is this not possible with the HAT?

Pybricks developer here.

We define stalled as "unable to reach target speed or position given the available feedback torque". The torque and speed thresholds are configurable. Using the internal motor control state, this can get [very responsive](https://www.youtube.com/watch?v=QmijLZzLjyk). 

On that note, would there be any interest in a _fully open source firmware_ for the Build Hat?

I've seen a few of such requests [here](https://github.com/RaspberryPiFoundation/python-build-hat/issues/132) and on the Raspberry Pi forums. It could potentially solve most [motor](https://github.com/RaspberryPiFoundation/python-build-hat/issues?q=is%3Aissue+motor+) and sensor related problems, and perhaps improve the hat's [responsiveness](https://www.youtube.com/watch?v=ZY8wgRCnZYQ&t=302s) quite a bit.

It's certainly possible technically, and we'd be happy to chat about it with the Raspberry Pi Foundation. I don't know if we should open an issue here (@chrisruk ? ) but we can also be reached via email (team@pybricks.com) or on [GitHub](https://github.com/pybricks/support/issues).

mutesplash, 2022-07-26:
I would say there's not enough political capital in online discussions with the Raspberry Pi Foundation to get any meaningful action on the firmware, as no reason for the closed source has been even offered, despite the chip being produced by the Foundation itself.  The Pybricks team may have enough, if the issue of BuiltHAT being not well suited for precision robot applications due to firmware limitations was raised in their own official channels.

I do like the BuildHAT, but I can't even get it to replace EV3Dev in some projects I have due to the imprecision in motor control.

laurensvalk, 2022-07-28:
Thanks for your response! But crucially, this approach doesn't require the original firmware source to be be disclosed.

Pybricks already has open source drivers for all LEGO motors and sensors. These drivers could be adapted to run on this architecture. They can run without MicroPython on top as this would not be needed in this application.

The Build Hat seems like it has a lot of potential just waiting to be unlocked :)

mutesplash, 2022-07-28:
The firmware is signed, so looping in Pybricks to run on the RP2040 would be a significant development structure change, if I understand you correctly.

If the firmware were open sourced, which seems to require external pressure on the Foundation to do so, then improved and adapted drivers could at least be formally entertained as options.

hollielucas, 2022-09-24:
I've come here as I have recently obtained a RPi 3 B and, being an AFOL who enjoys motorising builds, immediatly considered a build HAT. However, having read this thread, and others, I'll give it a miss. Consider:
Pyricks on the Technic Hub works.
Pybricks on the EV3 works.
EV3Basic on the EV3 works.
Arduino (with hacked wiring) works.
So many alternatives work as intended. Why put up with h'ware and s'ware that does not work.
So much good open source software out there that I have gained great pleasure, satisfation and knowledge from. Why this attitude by RPi Foundation?