RaspberryPiFoundation/python-build-hat #137: Inconsistent motor power

mutesplash, 2022-03-31:
Test program is as follows

```
import buildhat
import time

motor_a = buildhat.Motor('A')
motor_b = buildhat.Motor('B')
motor_c = buildhat.Motor('C')
motor_d = buildhat.Motor('D')

last_time = time.time()
motor_a.set_default_speed(10)
motor_b.set_default_speed(20)
motor_c.set_default_speed(25)
motor_d.set_default_speed(30)
motor_a.start()
motor_b.start()
motor_c.start()
motor_d.start()

time.sleep(20)

motor_a.stop()
motor_b.stop()
motor_c.stop()
motor_d.stop()
```

Observed problem is that at speed levels less than 30 (doesn't occur somewhere between 25 and 30) the motors (Medium Angular of both colors) are very jerky and do not rotate smoothly.  Large Angular Grey seems to exhibit this problem at speeds lower than 20.  They all creep very consistently when connected to Spike PRIME and set to 10 (but also emit a whine that they don't on BuildHAT).

The numbers are not at specific cutoffs because the motors seem to exhibit the problem at various levels depending on the speed at which they were last run, despite being stopped and the Python test program ending.

chrisruk, 2022-04-12:
Sorry for the delay, you could possibly have a look at trying different plimit/bias values such as like I did here - 
https://github.com/RaspberryPiFoundation/python-build-hat/issues/118#issuecomment-1043163318 , which helped for me at driving the lower speeds of the passive motor.

mutesplash, 2022-04-12:
Well, just by slapping those settings in, it worked, but shouldn't something be able to figure out which things to set automatically?

chrisruk, 2023-02-20:
I'm hoping the new firmware I'm testing should help with this, as a speed PID controller will be able to regulate speed by differentiating position.  There is also a new 'slower' PWM mode.

chrisruk, 2023-03-17:
I'm wondering if the new firmware might help with this, which you can try here -

```
pip uninstall buildhat
pip install git+https://github.com/RaspberryPiFoundation/python-build-hat.git@newfirmware2
```

I've also added the ability to specify speed in rpm, and tried measuring the speed across 4 motors running at the same speed using a laser tachometer.

```
m = Motor('A')
m.set_speed_unit_rpm(rpm=True)
m.start(speed=30) # Runs at 30 RPM
```

chrisruk, 2023-04-11:
I believe the motors function better at slower speeds now and also can achieve the same speed on different types of motor when using rpm mode, so will close this now.

If you notice any issues please feel free to re-open.

closed 2023-04-11 by chrisruk