RaspberryPiFoundation/python-build-hat #168: Lego Technic 69730 not hold position

Gadgeteering, 2022-10-23:
I want to use the Lego Technic 69730 motor as a servo, but once the 'run_to_position' command has finished, the motor is de-energised and is free to rotate to any position.

Is there away to hold a motor in a set position in a similar way as a servo ?

chrisruk, 2022-10-24:
Good point, yes this should be possible, I'll try and work on a pull request for this soon.

Gadgeteering, 2022-10-30:
I think it would need to be implemented in the firmware ?

chrisruk, 2022-10-31:
I just remembered there's already a variable called _release.  If you set it to false it will hold the motor position. 
For example -

```
from buildhat import Motor
m = Motor('A')
m._release = False
m.run_to_position(90)
```

Gadgeteering, 2022-10-31:
The test code above didn't work until I realised that once Python finishes running it releases the Hat and motor and it was free to rotate.

So I added an infinite loop and it showed that it works perfectly, Thanks

```
from buildhat import Motor
m = Motor('A')
m._release = False
m.run_to_position(90)
while True:
   None
```

chrisruk, 2022-10-31:
Ah yeah, good point about the loop.

I'm going to make a pull request that allows you use 'release' rather than '_release' and will document this

chrisruk, 2022-10-31:
Will re-open this issue, and close it again, when I've pushed the changes to do this.

Thanks again for reporting

chrisruk, 2022-11-01:
I've now made a public property for both Motor and MotorPair, called 'release' so you can do

```
m = Motor('A')
m.release = False
```

This will be part of the next release to PyPI.

closed 2022-11-01 by chrisruk