RaspberryPiFoundation/python-build-hat #169: Documentation - Type IDs

rjboonstra, 2022-10-24:
Regarding the documentation page located at https://www.raspberrypi.com/documentation/accessories/build-hat.html#device-compatibility
When I have two medium motors plugged in to A and B, they come up as Type ID 38 and not Type ID 1 as listed above.
_python lego03_hat.py
{'A': {'typeid': 38, 'connected': True, 'name': 'Motor', 'description': 'Medium Linear Motor'}, 'B': {'typeid': 38, 'connected': True, 'name': 'Motor', 'description': 'Medium Linear Motor'}, 'C': {'typeid': 2, 'connected': True, 'name': 'PassiveMotor', 'description': 'PassiveMotor'}, 'D': {'typeid': 35, 'connected': True, 'name': 'MotionSensor', 'description': 'MotionSensor'}}_

Also, on the github documentation, it might be helpful to include the Type IDs in brackets after the description to help know what is compatible with what - https://buildhat.readthedocs.io/en/latest/buildhat/index.html

My challenge was that I have a Motion Sensor (Type ID 38) and initially thought it was a Distance Sensor (Type ID 3E)... and did not realize this initially.  As a  side issue, any support for the Motion Sensor would be excellent (currently not supported).

mutesplash, 2022-10-24:
Punch the LEGO item numbers listed alongside the device IDs (38 & 35) from the RPi docs into bricklink.com and see if the visuals match what you have.

Also, sometimes the type IDs are listed places in integers and others in hex, and it's not clear which is which.  In this case, it's int 38, hex 0x26 for medium linear motor.  I have a note that RPi docs may have the ID incorrect for the Motion Sensor compared to the list at the top in devices.py.  I tried to put accurate LEGO item numbers in comments alongside each Type ID.

The MotionSensor currently will report the distance, what else would you like it to do?  Unfortunately I think that's the one LPF2 device that I don't have.  If you can post the initialization code ("connected to active ID") that the BuildHAT dumps out when you set the library to debug, that will tell us if there's another mode that may be interesting to expose.

chrisruk, 2022-10-24:
Mutesplash is correct in that table they're listed as hex, so 0x26 is what your motor is in that table.

Currently the MotionSensor can report the distance, by calling get_distance() -
https://github.com/RaspberryPiFoundation/python-build-hat/blob/main/buildhat/wedo.py

rjboonstra, 2022-10-25:
Thank you chrisruk / Mutesplash.  I did not realize that the MotionSensor function was available, but it worked just as it should:
Program:
_#!/usr/bin/python
import time

from buildhat import PassiveMotor
from buildhat import MotionSensor

motor = PassiveMotor('C')
sMotion = MotionSensor('D')

print("Starting Program...")
print("Distance", sMotion.get_distance())

while True:
        print("Start motor")
        motor.start()
        print("Distance", sMotion.get_distance())
        time.sleep(3)
        print("Stop motor")
        motor.stop()
        print("Distance", sMotion.get_distance())_

Output:
pi@rpi04:~/Documents $ python sensor01.py
_Starting Program...
Distance 10
Start motor
Distance 10
Stop motor
Distance 10
Start motor
Distance 10
Stop motor
Distance 10
Start motor
Distance 10
Stop motor
Distance 10
Start motor
Distance 10
Stop motor_

I am not sure of the right way to get the documentation updated, but MotionSensor is not listed:  [https://buildhat.readthedocs.io/en/latest/buildhat/index.html](https://buildhat.readthedocs.io/en/latest/buildhat/index.html
)
Thanks very much, Rudy

closed 2022-10-25 by rjboonstra