RaspberryPiFoundation/python-build-hat #174: high latency for Motor.get()

Zaspire, 2022-11-06:
```python
import datetime
from buildhat import Motor

motor = Motor('A')
while True:
  t = datetime.datetime.now()
  motor.get()
  print(datetime.datetime.now() - t)
```
Latency is between 150ms-200ms.

it's only possible to retrieve current position 5-6 times a second.
which makes hard controlling already running motor.

My use case: exactly controlling angle with joystick.

chrisruk, 2022-11-06:
That's a good point, this is because the 'selonce' command only selects one update from the motor.  If you'd like the motor to stream its updates the callback functionality should be able to provide a higher frequency of updates -

Could you have a look at the motor callback mechanism demonstrated in -
https://github.com/RaspberryPiFoundation/python-build-hat/blob/main/docs/buildhat/motor.py

Zaspire, 2022-11-07:
> callback functionality should be able to provide a higher frequency of updates

Thank you for suggestion.
callback mechanism is able to provide consistent updates every ~100ms(10hz).
But not enough to reliable position motor on half the speed.
40hz+ refresh will be awesome :)

chrisruk, 2022-11-08:
I believe to get a higher frequency of update, would require a firmware change.  I'll flag this up.

mutesplash, 2022-11-09:
Firmware already seems to have a hard time keeping up with its current frequency?

https://forums.raspberrypi.com/viewtopic.php?p=1963238#p1963273

chrisruk, 2022-11-23:
Apparently 100ms is part of LEGO's specification, so this can't be altered I'm afraid.

Zaspire, 2022-11-23:
> Apparently 100ms is part of LEGO's specification

Running following code on lego mindstorm hub gives different results:

```
import utime
import mindstorms

motor = mindstorms.Motor('A')

ms = utime.ticks_ms()
degrees = motor.get_degrees_counted()
while True:
  tdegrees = motor.get_degrees_counted()
  if tdegrees == degrees:
    continue
  print(utime.ticks_ms() - ms, tdegrees)

  ms = utime.ticks_ms()
  degrees = tdegrees
```

Output:
```
4 20606
6 20609
4 20612
6 20616
4 20620
6 20623
4 20626
6 20630
4 20633
6 20636
4 20639
6 20642
4 20646
6 20649
4 20653
```

5ms between updates is **200hz**.

chrisruk, 2022-11-24:
Thanks a lot! Will pass this information on to the firmware dev.

chrisruk, 2022-12-15:
For callbacks you can now alter the interval at which you get data.

There's a little example of how to do it here -
https://github.com/RaspberryPiFoundation/python-build-hat/blob/main/test/motors.py#L84

natikgadzhi, 2023-02-07:
I'm finally getting up to speed with a hobby project that uses buildhat, and will be happy to help as well. My usecase is going to be the same as @Zaspire, but with a twist that buildhat is wrapped in a grpc server that is streaming state to a client app, that has a gamepad paired to it. Quite a few spots to add latency. 

@chrisruk, if you'll need a pair of hands on this codebase to help implement any of our asks, I'm happy to try and get up to speed and help out.

chrisruk, 2023-02-07:
Cool, sounds fun!

I'm planning on soon altering the .get() calls so that they request a continuous stream of updates from the sensor/motor, rather than just requesting a single update, and taking advantage of 'selrate' to get much faster updates.

ArnavDagar, 2023-02-18:
That would be fantastic! Even with the callback method, my motors only give me 10 Hz update rate and I really need to use a 100 Hz get() rate. Do you have any ETA on this? Really appreciate it.

chrisruk, 2023-02-20:
I assume maybe you only got 10Hz with the callbacks, as the .interval value wasn't set to 10 (for 10ms or 100Hz).  I hope to have the changes for the .get() command too, finished over the next week or so.

chrisruk, 2023-02-24:
The changes for this are now in 'main' so will close this now.

closed 2023-02-24 by chrisruk