RaspberryPiFoundation/python-build-hat #175: Motor.run_to_position(blocking=False) does not check that no other calls are currently in progress

Zaspire, 2022-11-06:
Motor.run_to_position(blocking=False) starts new thread for each call.
Multiple calls with with in short time interval will result in multiple simultaneous calls to Device.get(), which blocks on ```Device._instance.portcond[self.port].wait()```
BuildHAT.loop will wake threads in random order.

It will be better, if each new call cancels all outstanding requests

chrisruk, 2022-11-06:
Thanks a lot for pointing this out.  I'll work on this soon, I was thinking I can possibly 
implement a queuing type mechanism so the threads are woken in the correct order.

chrisruk, 2023-02-03:
I've just committed to 'main' changes that preserve the ordering of non-blocking motor commands.

Also if you do for example:
```
        m = Motor('A')
        m.run_for_seconds(5, blocking=False)
        m.run_for_degrees(360)
```

The 'run_for_degrees' call, will wait for all non-blocking calls to finish, prior to running.

natikgadzhi, 2023-02-07:
@chrisruk, thank you for fixing this! 

On a similar note, let's say I'm using a motor for steering and accepting and queueing remote commands. If I queue a command `run_for_degrees` and it's not yet started executing, but another command `run_for_degrees` in different direction comes in, I should be able to go to `hat.motorqueue[target_port].pop` or delete another command from queue, right? 

That's unless that command is currently executing. If it's already in progress, any way to see what command is in progress on each port, and interrupt it if needed?

chrisruk, 2023-02-07:
You'd have to find the correct run_for_degrees command in the queue which you'd want to remove, which could be tricky as
the synchronized queue class doesn't publicly expose the ability to remove specific elements as far as I'm aware, apart from grabbing and removing the first element, via .get().

If a command is currently in progress, currently you wouldn't be able to interrupt it.  That's an interesting question though.

chrisruk, 2023-02-09:
Will close this issue now, as threads should now be woken in the correct order, due to the use of a queue system.

closed 2023-02-09 by chrisruk