RaspberryPiFoundation/python-build-hat #184: lights (ID 8) cannot be turned off completely

arohrbacher, 2023-01-09:
Hi all,

running this python code
`
from buildhat import Light
light = Light('A')
light.on()
light.off()
`
doesn't turn off the lights completely - they stay on dimly lighting.
Also light.brightness(0) won't turn them off completely.

Am I missing something? I am running the latest version of the buildhat python library (0.5.12)

Thanks, Andreas

chrisruk, 2023-01-12:
Hi,

I'm afraid I can't replicate that with my 88005 lights ( https://www.lego.com/en-gb/product/light-88005 ), .off()
turns them completely off for me.

Do your lights appear the same as in the lego photo?

I assume this happens regardless of which port you attach them too?

Sorry I can't be of more help.

arohrbacher, 2023-01-12:
Thanks for answering. Yes, this issue happens on all 4 ports.

However, I don't have the original LEGO light - I made the lights myself: I connected two white LEDs (with series resistors) to a plug and also connected the 2.2kOhm resistor for identification to the proper pins. So it seems that the original LEGO lights are electrically a bit different. My homemade lights function properly (i.e. turn off completely) on a LEGO city hub or LEGO technic hub, though.

I found a workaround for BuildHat, though: if you add this 
`        if (brightness==0):`
`            self._write(f"port {self.port} ; coast ; plimit {brightness / 100.0}\r")`
to `light.py`, the lights turn off completely when you do
`light.brightness(0)`

Note the `coast` command - this seems to switch off the drivers. If this is not issued, the drivers stay partially on for some reason...

chrisruk, 2023-01-13:
Thanks for the reply, I wonder if the forward voltage of the LEDs used could be different, or the series resistors are different values, to that of the LEGO lights.

arohrbacher, 2023-01-13:
My resistors are 1.5 kOhms each for two white LEDs in parallel; I also have diodes to protect them from reverse voltage, so it is possible that the total forward voltage of my setup is different from the LEGO lights. Using the `coast` command ensures that the drivers are completely switched off, so I am using it in my patched `light.py`. Maybe this change helps others wanting to connect homemade LED lights to the BuildHAT...

chrisruk, 2023-01-13:
I'll double check with the firmware dev, to make sure 'coast' doesn't have any downsides, instead of 'off', if not, will look at changing this soon, for at least the Light class

chrisruk, 2023-02-02:
I've switched to using coast now, so your own lights should turn off completely.  Thanks for the report.

closed 2023-02-02 by chrisruk