RaspberryPiFoundation/python-build-hat #185: Is it possible to control a lego power function motor with the build hat?

HannahFehringer, 2023-01-11:
Hello,

basically my issue title already sums up my question: Is it possible to control a lego power function motor with the build hat?

There do exist some cable which allow a connection between powered up and power function devices. However, if I connect my lego pf motor with the raspberry pi build hat, I get the following error message: 
`buildhat.exc.DeviceError: There is not a Motor connected to port A (Found PassiveMotor)`

This tells me that it found a motor, but not the expected one. Does anybody have experience with this issue?

mutesplash, 2023-01-11:
You need to use the PassiveMotor class and not the Motor class.

chrisruk, 2023-01-12:
Yes, that's correct, there's an example at the bottom of https://buildhat.readthedocs.io/en/latest/buildhat/passivemotor.html

chrisruk, 2023-01-13:
Will close this now, feel free to open another issue if there's anything else

closed 2023-01-13 by chrisruk