RaspberryPiFoundation/python-build-hat #187: Running in Virtual Environment

dhruvsheth-ai, 2023-01-13:
I can run Buildhat programs without a virtual environment but a project I'm working on requires most libraries to be in a virtual environment to avoid conflict with the standard environment. Is there anyway I can run the code on a virtual environment for buildhat?
Model: Rpi 4 8gb ram
32bit bullseye
32gb storage

The error:
```shell
Traceback (most recent call last):
  File "/home/raspberry/depthai-python/examples/SpatialDetection/spatial_mobilenet_lego.py", line 14, in <module>
    motor2 = Motor('B')
  File "/home/raspberry/venv3/lib/python3.9/site-packages/buildhat/motors.py", line 108, in __init__
    super().__init__(port)
  File "/home/raspberry/venv3/lib/python3.9/site-packages/buildhat/devices.py", line 57, in __init__
    Device._setup()
  File "/home/raspberry/venv3/lib/python3.9/site-packages/buildhat/devices.py", line 80, in _setup
    Device._instance = BuildHAT(firm, sig, v, **kwargs)
  File "/home/raspberry/venv3/lib/python3.9/site-packages/buildhat/serinterface.py", line 146, in __init__
    raise BuildHATError("HAT not found")
buildhat.exc.BuildHATError: HAT not found
```

Thank you!

dhruvsheth-ai, 2023-01-13:
For those stuck with this error, it works when you open a new terminal to run the same code in standard environment and then returning to the terminal with the virtual environment and rerunning the code.

closed 2023-01-13 by dhruvsheth-ai

chrisruk, 2023-01-13:
With a venv, you should just be able to do something like:

```
python3.9 -m venv myenv  # to create venv
source myenv/bin/activate  # to activate 
pip3 install buildhat             # to install buildhat
```