RaspberryPiFoundation/python-build-hat #191: Color Sensor Sampling Rate of only 2 Hz

ArnavDagar, 2023-02-04:
``Hi,

The color sensor sampling rate is only 2 Hz, though the actual rate on the Spike Prime Website is 100 Hz (https://education.lego.com/v3/assets/blt293eea581807678a/blt62a78c227edef070/5f8801b9a302dc0d859a732b/techspecs_techniccolorsensor.pdf).

`from buildhat import Motor, ColorSensor
import time

color_sensor = ColorSensor('B')

for i in range(10):
    print(time.time(), color_sensor.get_color())
`

Output: 
1675486816.6453307 black
1675486817.074805 black
1675486817.4981308 black
1675486817.935994 black
1675486818.3843992 black
1675486818.821309 black
1675486819.2702649 black
1675486819.7176251 black
1675486820.1744425 black
1675486820.6134553 black.

Thanks in advance. 
Arnav

mutesplash, 2023-02-06:
BuildHAT is slow.  Issue #174 has another thread about update speeds for a different device.

chrisruk, 2023-02-07:
One thing which may slightly help currently is to do -

```
color_sensor.avg_reads = 1
```

I'm going to try altering the library so that when you call ```color_sensor.get_color()``` it starts continuously obtaining values from the sensor at a user specified rate, which could be 100Hz etc, rather than just requesting a single update.

chrisruk, 2023-02-09:
I've started working on a new pull request for this and have been able to get >= 100Hz, I need to do more testing with multiple devices though.

ArnavDagar, 2023-02-18:
Thank you,

I want to experiment with this - how do I update my installed library to use the code? Do I need to reinstall the library?

chrisruk, 2023-02-20:
If you want to try this, you'd have to checkout the sensorinterval branch (https://github.com/RaspberryPiFoundation/python-build-hat/tree/sensorinterval), you'd need to 'drop' the commit that says 'Drop this commit before release' though.

I'll hopefully try and get the changes from this branch, merged soon though into 'main'.

chrisruk, 2023-02-24:
The latest changes are now in 'main' to try you can do the following -
```
pip uninstall buildhat
pip install git+https://github.com/RaspberryPiFoundation/python-build-hat.git
```

You might want to do the following to remove the 'averaging code'

```
color_sensor.avg_reads = 1
```

closed 2023-02-24 by chrisruk

ArnavDagar, 2023-02-26:
Hi @chrisruk,

Thank you so much for setting this up - I tested this out and it works great. 

However, I am also using a RPLidar A1 with a UART USB Port. When I use the buildhat, it ends up choking the RPLidar.

I need to revert to the buildhat library that I was using from December, which was working earlier. How would I go about doing this?

Thanks,
Arnav