RaspberryPiFoundation/python-build-hat #194: ValueError when Logging Data w/ 100 Hz Rate

ArnavDagar, 2023-02-26:
Hi,

With the new changes, I found that when I try to log motor data, I receive the following error after around 30 seconds to a minute:

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/buildhat/serinterface.py", line 360, in loop
    newdata.append(int(d))
ValueError: invalid literal for int() with base 10: '+2\x009'

In another test, I got:
1. ValueError: invalid literal for int() with base 10: '+4\x009041'
2. ValueError: invalid literal for int() with base 10: '\x00'

The new changes also choke my RP Lidar Connected to USB port through a Serial to USB adapter for UART.

Can you please let me know how I can revert to the old library from December that worked much better for me. I am new to github and need help to go to an older library

Thank you so much,
Arnav

chrisruk, 2023-02-26:
I'm a bit confused about the error you're getting, I've been running the following for around 10mins without issue.

```
#!/usr/bin/python3.9
import time
from buildhat import Motor, Hat

h = Hat(debug=True)
m = Motor('B')

m.start(20)

while True:
    print(m.get())
```

I wasn't sure what you meant by logging.  Please can you paste the python code you were running.

The line number in your error message also confused me as https://github.com/RaspberryPiFoundation/python-build-hat/blob/main/buildhat/serinterface.py#L360 line 360 doesn't seem to be what is in 'main' currently.

ArnavDagar, 2023-02-26:
Hi Thanks, I will try, but right now I am very short on time for a deadline. Can you please guide me how to go to an older library from Dec that was functioning with my code.  I am new to github and need some handholding.

In my project, besides the buildhat for Lego components, I am using a RPLidar A1 and a TFLuna which use UART over serial to USB interface with the raspberry pi. 
The RPLidar does not even function with the new library as it runs into buffer issues. And the TFluna based project runs into the value error issue 
Both these projects were running fine (but with lower motor and color sensor rates) before I went to the new library.

My project is due this Friday and I need to go to the older library for now. Can you please let me know how?

Thank you so much

Arnav

ArnavDagar, 2023-02-26:
Thanks! This will take me to the library from Dec or the one with the updated motor and color sensor rates?

chrisruk, 2023-02-27:
It will install the latest version from the PyPi repository which was released in December ( https://pypi.org/project/buildhat/ ).

chrisruk, 2023-03-02:
Would you be able to show the snippet of python that triggers the issue you're referring to?  It's not clear to me what you mean here by logging.

At the moment I can't replicate this, apart from by opening a terminal emulator on the same serial interface.

Thanks!

closed 2023-03-16 by chrisruk