RaspberryPiFoundation/python-build-hat #197: Motors keeps momentum after it is stopped

williamcorsel, 2023-04-07:
I noticed that motors seem to keep some momentum after they are stopped and started again when using the current `main` branch. There also seems to be quite a long spin-up time in this case. 

An example script to simulate the behaviour:

```python
from buildhat import Motor
import time

speed = 50
delay = 4
delay_stop = 2

motor = Motor('A')

while True:
    print(f"Start {speed}")
    motor.start(speed)
    time.sleep(delay)

    print("Stop")
    motor.stop()
    time.sleep(delay_stop)

    print(f"Start {-speed}")
    motor.start(-speed)
    time.sleep(delay)

    print("Stop")
    motor.stop()
    time.sleep(delay_stop)
```

Motor behaviour:

https://user-images.githubusercontent.com/31770711/230624586-2d7e3ba3-9233-42b2-8c76-d8ee5be5a8da.mp4

Are there some settings I can tweak to improve the responsiveness in this case? I noticed that using `motor.set_speed_unit_rpm(True)` already reduces the momentum issue somewhat (and improves the spin-up time significantly).

chrisruk, 2023-04-13:
Thanks for reporting this and especially for the video, that's very helpful!

danielr, 2024-01-04:
I can confirm the exact same behavior, which makes controlling a robot almost impossible. 😟 Did you find any solution or workaround?

williamcorsel, 2024-01-05:
Sadly no. You could try and use the `.pwm` [function](https://github.com/RaspberryPiFoundation/python-build-hat/blob/78c0c6988f9c786113b12c4c6309aecccb36b1da/buildhat/motors.py#L441) directly.