RaspberryPiFoundation/python-build-hat #198: Program freezes after run_to_position command

schultz77, 2023-08-10:
Hi there,

i've implemented a simple python script with a board camera for Pan/Tilt Tracking Control.
I am using two motors for tracking an object. For move to the desired position i am using x_motor.run_to_position(x) and y_motor.run_to_position(y). After a while the program freezes.
These are the terminal messages after a keyboard interruption  

_File "/usr/local/lib/python3.9/dist-packages/buildhat/motors.py", line 210, in _run_positional_ramp
    self._hat.rampcond[self.port].wait()
  File "/usr/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt_

Thx in advance for your support!

code excerpt (inside a while loop):

`if Track == 1:
                # calculating the center of gravity of contour
                if cog_contour_x:
                    cog_contour_x =  0.8*cog_contour_x + 0.2*(x+w/2) # low-pass filter
                else:
                     cog_contour_x = x+w/2
                     
                if cog_contour_y:
                     cog_contour_y = 0.8*cog_contour_y + 0.2*(y+h/2) # low-pass filter
                else:
                     cog_contour_y = y+h/2

                # calculating distance between cog_contour and the center of the window
                error_x = cog_contour_x - dispW/2
                panAngle += error_x/PIXEL2DEGREE_RATE
               
                if panAngle > MAX_PAN_ANGLE:
                    panAngle = MAX_PAN_ANGLE
                if panAngle < -MAX_PAN_ANGLE:
                    panAngle = -MAX_PAN_ANGLE
                
                if abs(error_x) > MIN_PIXEL_ERROR:
                   x_motor.run_to_position(panAngle)
                
                error_y = cog_contour_y - dispH/2
                tiltAngle -= error_y/PIXEL2DEGREE_RATE
                                                
                if tiltAngle > MAX_TILT_ANGLE:
                    tiltAngle = MAX_TILT_ANGLE
                if tiltAngle < -MAX_TILT_ANGLE:
                    tiltAngle=-MAX_TILT_ANGLE
                                
                if abs(error_y) > MIN_PIXEL_ERROR:
                   y_motor.run_to_position(tiltAngle)`

schultz77, 2023-08-12:
Hi there again,

i've observed a further behavior of my motor. Sometimes i've stoped the program without freezes. Then i've got following terminal messages:
_serial.serialutil.PortNotOpenError: Attempting to use a port that is not open_

Am i stressing the motor to much?

mutesplash, 2023-08-12:
The serial error is from the serial port connection to the BuildHAT.  My assumption is that you are sending commands too fast to the BuildHAT, it's getting overloaded and acting strange.

schultz77, 2023-08-13:
Thank you very much for your reply!
I've tried to slow down the frequency of updating the motor position. It became better, but object tracking with build hat seems to be too slow. I am now trying with 2 servos. It seems to be a bit more robust.

schultz77, 2023-09-08:
see comments above

closed 2023-09-08 by schultz77