RaspberryPiFoundation/python-build-hat #200: running motor simultaneously

schultz77, 2023-09-08:
Hi there,

I'am trying to run four lego motors with raspberry pi 4 and build-hat simultaneously.
Using MotorPair it worked well for 2 motors. Is there any possibility to run 2 motor pairs simultaneously?

This is probably not a bug but a feature.
Thx in advance!

mutesplash, 2023-09-08:
Have you tried it?

schultz77, 2023-09-08:
Hi mutesplash,

When I  execute run_to_position for one motor pair (A, B) and then for the other one (C, D)  they run after each other. First A. B then C, D.
I would like to have a motor quadruple A, B, C, D.

Is it somehow possible and I misunderstood the documentation? 

Thank you!

mutesplash, 2023-09-08:
Have you specified to run them nonblocking?

schultz77, 2023-09-08:
Thank you for your quick reply!
I'll try it and give you feedback at the latest tomorrow!

schultz77, 2023-09-09:
Hey mutesplash,

thank You very much. It works!

schultz77, 2023-09-09:
see comments above

closed 2023-09-09 by schultz77