RaspberryPiFoundation/python-build-hat #202: Support for sending IR signals from colordistancesensor

kebe67, 2023-09-14:
As far as I am able to see, the mode for sending IR signals from the colordistance sensor to the older Power Functions receivers is not supported in the library. Would it be possible to get support for this in some way?

mutesplash, 2023-09-15:
That looks like IR Tx Mode 7.  You don't happen to have any documentation on that or part numbers it's supposed to interop with?

mutesplash, 2023-09-15:
https://www.philohome.com/pf/LEGO_Power_Functions_RC_v120.pdf

Hm, it looks like maybe you stuff everything between start and stop into two bytes and send that.  Unfortunately I have none of the LPF (LPF1?) devices to just throw junk at it and see what happens.

mutesplash, 2023-09-26:
I'm just an enthusiast but @chrisruk works for RPi.  I haven't found a three-piece LPF1 testing setup for well under $100 on bricklink yet, though.

If you're comfortable testing out a fork, I can work up something to set those 12 bits and you can run it and see what happens.  I can only make sure the sensor isn't reporting errors when I send the data to it.

mutesplash, 2023-09-26:
Or maybe he did 😅 , it's been a while!

chrisruk, 2023-10-07:
@kebe67 I'm afraid I don't have a power function receiver to test reception with.

@mutesplash yeah I have moved to another company.

mutesplash, 2023-11-19:
@kebe67 I borrowed some hardware to test this out soon

mutesplash, 2023-11-28:
@kebe67 Got the motor to move!  So it's possible, I just don't quite have it figured out

mutesplash, 2023-11-29:
I have combo PWM mode working reliably on what appears to be V1 hardware on the receiver with the 6116239 battery box driving motor 6000564.  Mode 7's format details that it takes 5 nibbles, the first two taken up by the default header for `write1()` specifying which mode (7) to write to (0xC7).  The nibbles seem to be sent in inverted order, with nibble 1 being last, padded by zeros in the upper bits.  You don't have to calculate LRC; BuildHAT or the sensor itself seems to do that for you.

[Philo](https://www.philohome.com/pfrec/pfrec.htm) noted that there's a bug in V1, which would explain why trying to follow the documentation produced maddening results, especially since I tried to implement that mode first.  I will see if a friend has V2 in their extensive library of parts.

I want to get a few more modes working before I submit a PR.

mutesplash, 2023-12-14:
@grega Thoughts on getting this merged?

grega, 2023-12-19:
@mutesplash done 👍 thanks for your work on this!

I'll publish a new release shortly...

closed 2023-12-19 by grega