RaspberryPiFoundation/python-build-hat #207: How is the firmware written?

Eisverygoodletter, 2023-12-19:
How is the file
https://github.com/RaspberryPiFoundation/python-build-hat/blob/main/buildhat/data/firmware.bin
created and how would I go about writing my own firmware? 
I understand that this file is loaded into the build hat by `serinterface.py` every time the build hat is loaded, but I don't know how this firmware file could be created. 
I'm experimenting with using a non-official sensor that doesn't have an ID code specified by the firmware documentation at https://datasheets.raspberrypi.com/build-hat/build-hat-serial-protocol.pdf

mutesplash, 2023-12-19:
That would be real nice but as far as I can tell the firmware is signed by developers at the Raspberry Pi company and _presumably_ the RP2040 is set to not load any third party firmware.  It would be a huge upgrade to get the firmware on github, even if we couldn't do anything with it, for debugging and reasonable feature requests.

grega, 2023-12-19:
We are aware of these issues, but not currently able to make any promises or commitments. Will update if that changes.

mutesplash, 2023-12-19:
But is Raspberry Pi... _interested_ in resolving the issue?

grega, 2023-12-19:
Yes, but it is potentially complex. Am hoping to be able to explore the matter more over the next couple of months.

Eisverygoodletter, 2023-12-25:
Thanks for the response! I might settle on just using the official sensors for now.

closed 2023-12-25 by Eisverygoodletter

mutesplash, 2024-04-18:
@grega Any new information on this front?

grega, 2024-12-02:
We're expecting to have some news on this front around mid-January. Apologies for the delays!

mutesplash, 2024-12-02:
News!  Thank you!

laurensvalk, 2024-12-25:
> Yes, but it is potentially complex. Am hoping to be able to explore the matter more over the next couple of months.
> 
> We're expecting to have some news on this front around mid-January. Apologies for the delays!

Over at Pybricks we'd still be happy to make a community-based open source firmware replacement using the Pybricks device drivers, which covers all known LEGO motors and sensors.

We _shouldn't need any of the original proprietary source code_ for the hat -- just a way to sign binaries for compatibility with the hat's bootloader. If the Raspberry Pi Foundation team wants to get in touch, please feel free to reach out at team@pybricks.com.

grega, 2024-12-27:
@laurensvalk yep was planning on getting in touch about this soon 👍 am almost certain we'll be able to do something here in the not-too-distant future...

grega, 2025-03-10:
Imminent. Waiting on a couple of factors not in my direct control but hoping it won't be more than a couple of weeks...

vodorok, 2025-04-29:
Hi @grega, do you have any updates?

tlaguz, 2025-06-09:
Hello @grega, are there any updates?

AndyA, 2025-06-30:
Hi @grega - any update on this? Thanks.

grega, 2025-07-09:
@mutesplash glad to hear! Really pleased to have been able to make this happen finally, and I hope that it will open up many more opportunities for everyone...

AndyA, 2025-07-14:
Fantastic news. Thank you @grega

vodorok, 2025-07-31:
Hi @grega, kudos to you for not forgetting the conversation in this obscure issue. These are great news!