RaspberryPiFoundation/python-build-hat #210: Python Build Hat for Ubuntu 22

wmbbarnett, 2024-01-27:
Is there a port of the Python Build Hat libraries for Ubuntu 22?

lurch, 2024-01-28:
According to the Ubuntu package-search, it's only available for `mantic` and `noble` https://packages.ubuntu.com/mantic/python3-build-hat
( https://wiki.ubuntu.com/DevelopmentCodeNames says Mantic Minotaur is 23.10 )

wmbbarnett, 2024-01-28:
Ubuntu 22 is the Long Term Support version, supported until 2032.  ROS 2 is available only on Ubuntu 22.    Since I am building a project based on robotics, I need this infrastructure.  Not having Build Hat for Ubuntu 22, dictates that I must move to a different motor control environment.

lurch, 2024-01-28:
Do the instructions [here](https://github.com/RaspberryPiFoundation/python-build-hat/blob/main/README.md#installation) not work on Ubuntu 22 ?

wmbbarnett, 2024-01-29:
traceback from the example:
 File "/usr/src/car/build-hat-test.py", line 5, in <module>
    motor = Motor('A')
  File "/home/pi/.local/lib/python3.10/site-packages/buildhat/motors.py", line 108, in __init__
    super().__init__(port)
  File "/home/pi/.local/lib/python3.10/site-packages/buildhat/devices.py", line 58, in __init__
    Device._setup()
  File "/home/pi/.local/lib/python3.10/site-packages/buildhat/devices.py", line 82, in _setup
    Device._instance = BuildHAT(firm, sig, v, **kwargs)
  File "/home/pi/.local/lib/python3.10/site-packages/buildhat/serinterface.py", line 157, in __init__
    raise BuildHATError("HAT not found")
buildhat.exc.BuildHATError: HAT not found
 

Note that I add to $PATH /home/pi/.local/bin and changed the permissions on /dev/ttyS0

wmbbarnett, 2024-01-29:
ran the sample twice, waited two minutes for response.  Each time it hung on line 5   "motor = Motor('A')"  traceback follows
 
Traceback (most recent call last):
  File "/usr/src/car/build-hat-test.py", line 5, in <module>
    motor = Motor('A')
  File "/home/pi/.local/lib/python3.10/site-packages/buildhat/motors.py", line 108, in __init__
    super().__init__(port)
  File "/home/pi/.local/lib/python3.10/site-packages/buildhat/devices.py", line 58, in __init__
    Device._setup()
  File "/home/pi/.local/lib/python3.10/site-packages/buildhat/devices.py", line 82, in _setup
    Device._instance = BuildHAT(firm, sig, v, **kwargs)
  File "/home/pi/.local/lib/python3.10/site-packages/buildhat/serinterface.py", line 118, in __init__
    self.write(b"version\r")
  File "/home/pi/.local/lib/python3.10/site-packages/buildhat/serinterface.py", line 257, in write
    self.ser.write(data)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 640, in write
    abort, ready, _ = select.select([self.pipe_abort_write_r], [self.fd], [], None)

mutesplash, 2024-01-29:
Your code is getting hung on the serial port code.  I would try communicating directly with BuildHAT with a terminal emulator to determine if you have the serial port set up properly.  If it is working, you may need to pass the device to use into the HAT constructor if it is not the default that the library expects.

lurch, 2024-01-29:
> have the serial port set up properly

Here's how you'd do that in Raspberry Pi OS, but I've not used Ubuntu on Pi so I dunno what steps would be needed for that 
https://www.raspberrypi.com/documentation/accessories/build-hat.html#setting-up-your-raspberry-pi

wmbbarnett, 2024-01-29:
Many thanks.  The serial port was the issue.  
 I edited cmdline.txt in  /boot/firmware to remove 'console=serial0,115200'
 config.txt already had the uart enabled.

I rebooted and my various devices began working.  No need to switch my motors!!!

closed 2024-01-29 by wmbbarnett