RaspberryPiFoundation/python-build-hat #211: Motor remains slow

vliedel, 2024-02-20:
I have a technic large motor connected to the build hat, and tried a bit of example code:
```
from buildhat import Hat
from buildhat import Motor

def handle_motor(speed, pos, apos):
    """Motor data

    :param speed: Speed of motor
    :param pos: Position of motor
    :param apos: Absolute position of motor
    """
    print("Motor", speed, pos, apos, "Vin", hat.get_vin())

hat = Hat()
print(hat.get())
print(hat.get_vin())
hat.set_leds(color='voltage')

motor = Motor('A')
motor.when_rotated = handle_motor
motor.run_for_seconds(10, speed=-100)
```

The motor doesn't go over speed 30, and around 25 most of the time.
The Vin stays above 7.8V.

I tried using `set_speed_unit_rpm(True)` as well, but no change in speed.

LukasFrinn, 2024-03-20:
@vliedel you need to set motor power limit.
motor.plimit(1)

vliedel, 2024-03-27:
Thanks! With `motor.plimit(1)` and `set_speed_unit_rpm(True)`, I finally reached speed 100 :)

closed 2024-03-27 by vliedel