RaspberryPiFoundation/python-build-hat #213: Question about PWM and motor control with Raspberry Pi 4B and Lego Build HAT

PinkPantherPC, 2024-08-15:
Hello everyone,

I am planning a project using a Raspberry Pi 4B and the Lego Build HAT to control motors and sensors. However, I have read that without using the pwm function, communication with the motors can be slow. 

I got the information from another issue:
- [Motors not starting at same time](https://github.com/RaspberryPiFoundation/python-build-hat/issues/152)

Could someone clarify to what extent this slowness affects motor performance? Specifically:

- Do the motors start with a noticeable delay?
- Do the motors start at significantly different times?
- Are there any issues with sensor communication, or does that work without any problems?

Any insights or experiences you could share would be greatly appreciated.

Thank you for your help!

Best regards,
PinkPantherPC

mutesplash, 2024-08-15:
What's "noticeable" and "significant"?

PinkPantherPC, 2024-08-16:
By "noticeable," I mean whether there is a perceptible delay between when the command is issued in the program and when the motor actually starts running.

By "significant," I am referring to whether, when using `MotorPair` to start two motors simultaneously, the robot moves straight or veers off course due to differences in the starting times of the two motors.

Thank you for your clarification!

gregorianrants, 2024-10-10:
I have written an alternative library that overcomes this limitation of the official library.

you can find it here
https://github.com/gregorianrants/buildhat-alternative