RaspberryPiFoundation/python-build-hat #214: Cannot select mode number >= 8

muzkr, 2024-12-15:
For the ultrasonic distance sensor (device ID 0x3E), there are 9 modes (numbered 0 - 8). After sending the serial command `select 8`, the BH does not return any data.
Although, it is possible to indirectly get the data for mode 8 by configuring a combi, e.g. `combi 0 8 0 8 1 8 2 8 3; select 0`, it seems that there is a bug in the implementation of the BH fw for reading the data of modes numbered >= 8.
I have no direct evidence, but would suggest that developers check to see if any leading CMD_EXT_MODE message is ignored before reading the MESSAGE_DATA message.

mutesplash, 2024-12-16:
Mode 8 is a calibration mode that does not specify accepting input or output in the port information?

muzkr, 2024-12-16:
> Mode 8 is a calibration mode that does not specify accepting input or output in the port information?

I don't known if the name of a mode or the actual meaning of its data would justify not returning it. Mode 5 of the medium motor (device ID 0x4B) in the Robot Inventor 51515 set is also CALIB, and BH returns data as expected upon input of command `select 5`.