RaspberryPiFoundation/python-build-hat #215: Confused by the naming convention on the device listed mode, and how to use combo modes

galop1n, 2025-02-23:
I am toying with the build hat with the intent to write a rust crate for it rather than using python. I got the basic right by uploading the firmware and signature and have a small REPL loop to toy with the interface and now trying to make sense of all I see.

Currently focusing on the light sensor.  In all modes I see, they state `SI = TYPE`. What does SI even means ? And PCT ? Especially given that the min max reported have both PCT, SI and RAW ( so what does M2 AMBI SI = PCT even means ? )

Is there a finite list of type ? So far I encountered IDX, PCT, RAW, DEG, CAL and MIN.

Next question is how combo works ? When the sensor report `C0: M0+M1+M5+M6`, is it like a shortcut to ask the sensor to spill all mode ? Is it related to the combi command ?

For the 3x3 light matrix. how to do know the write1 byte header value ? The example always goes `select X ; write CX p...`. switching mode also reset the light matrix state, so how does the fade animation of mode 3 can work ?

And last for now, Is the firmware open source ? and what is the signature for ?

Thanks

mutesplash, 2025-02-23:
The mode information is reported by the device.  SI for Système international (whatever SI units the device measures in).
PCT for percentage.
The type is also reported by the device, could also be anything.  There's LOAD, POS, and SPEED for motors.
Combo is indeed a shortcut.  Combi combo, same sort of thing.
How do we know the matrix data?  For one, reading the documentation for BuildHAT.  Two, experimentation.  Three, hints given by the [Lego BLE documentation](https://lego.github.io/lego-ble-wireless-protocol-docs/)  If you want the fade animation, use matrix.set_transition() then set the pixels you want to fade in.
The firmware is not open source.  There was supposed to be some communication last month about if that status would change, but it has not happened.  [muzkr / hachi](https://github.com/muzkr/hachi) may explain how the signature works, because it bypasses it...

galop1n, 2025-02-24:
Oh, shame on my French blood me on not figuring what SI was :) As for PCT, I would have not guess either, it took me years to realize what "X-ing" printed on the road meant "crossing" for my defense...

For the transition, I figured after posting that it actually worked once set, it is just a bit disappointing that the LED state reset tho. Once it will be set programmatically, i can use a shadow state to restore it, it may not be visible at all or a slight flicker, I will see.  

And yes, I am also digging the python library to work out some details, like the expected checksum of the firmware.

Thanks for the hachi link, will look at that also

mutesplash, 2025-02-24:
Flicker?  There are some notes in matrix.py about how to manage the transitions.