RaspberryPiFoundation/python-build-hat #216: Triangle profile

salbertini, 2025-03-30:
Hello, 
I'm trying to set a triangle profile with build hat and python library, but no way.
If I use a serial interface and follow build-hat-serial-protocol, everything works well. Different story with python. 
Another big issue with python is that the communication doesn't react as a serial interface.
I'm using build Hat on a raspberry pi 5 and the PI OS at this version:
Operating System: Debian GNU/Linux 12 (bookworm)
          Kernel: Linux 6.6.74+rpt-rpi-2712

Do you have some more example about motor control with profile in python?
Do you have a working python file that communicate with build hat in a good way?

Thanks
Stefano

mutesplash, 2025-03-31:
Got an example of the command you're using on the serial port and the function call you're trying to get work the same in Python?

salbertini, 2025-03-31:
Thanks mutesplash,
here a screenshot of the serial interface on PI5:

[image]

I've sent "help" command, and I received an answer. 
Please look at "CR" configuration that I've used, otherwise not work.

here my simple python code 

`import serial
from time import sleep
import serial

ser = serial.Serial("/dev/serial0",115200,timeout=1)

ser.write(f'help \r'.encode())

for i in range(100):
    line = ser.readline()
    line = line.strip()
    print(line.decode())
`

I don't have any errors, in vscode terminal  the cursor move down but no letters or other characters.

Please let me know if I'm wronging something and if you can propose some python code to test on my side.
Thanks
Stefano

mutesplash, 2025-04-09:
Sorry, thought you were actually trying to set the triangle motor profile, not do basic serial interaction.  You can check serinterface.py in the repo to see what's messed up:
* Not sending binary
* strip then decode vs decode then strip